goopypanther/hackerboat #33: Database reads interfere with writes

logos-electromechanical, 2016-07-13:
When we have a single writer to a DB and multiple readers, the readers do not release the DB correctly when they are done, causing all future write attempts to fail with error code (5) SQLITE_BUSY. This appears to be a failure to correctly release the DB lock at the end of the database access.

TTLComputer, 2016-07-13:
I think its hackerboatStateStorage::databaseConnection only ever requesting read+write access. It probably needs to be changed to readonly by default, and take an extra argument indicating that the consumer wants write access. That way it's hard to accidentally request write access to the db.

logos-electromechanical, 2016-07-13:
I think that's correct, but it's worse. The first time you call <object>->storage(), it opens a database connection and leaves it nailed up as long as the process is running; any subsequent calls to that function just return a pointer to that connection. It also calls createTable() when it opens the connection, and I suspect that's time consuming enough that we need to do something a bit more clever than just modifying the various places in sqliteStorage.cpp that call storage().

logos-electromechanical, 2016-07-13:
The other issue is that a single frame might call several different operations against the database, so it would make sense to open it, do shit, then close it at the calling level. I *think* the solution is going to be something like adding a close() function to hackerboatStateClassStorable. The original storage() function is protected, so I don't know how clean that will be from an interface perspective. Kind of ugly, really.

logos-electromechanical, 2016-07-13:
There should be separate read and write opening functions, but I'm going to write that up as an enhancement issue once this is fixed.

TTLComputer, 2016-07-13:
Generally, you want to leave DB connections open, as setting them up is expensive, though they should be closed when hackerboatStateClassStorable is destroyed. The main thing to do is when opening the connection is to specify if you want write access. createTable() should be reasonably quick (get a few ms), but requires write access so it will only need to be called if the code needs to write (if read only, we could check for existence of the table, then wait and report error after a timeout if it not there)

logos-electromechanical, 2016-07-13:
SQLite3 is server-less... it is just opening a disk file. I don't see why it would be costly. What am I missing?

TTLComputer, 2016-07-13:
It is mostly a file open, but it also does a few reads from the file. Also, if we wanted to switch to another SQL DB in the future, assuming they are cheap will be hard to undo. Debugging can be easier if the connection stays open, rather than closing the instant its done (although C++ doesn't really let you run arbitrary statements in the debugger while stopped at a break-point).

logos-electromechanical, 2016-07-14:
There's at least one place where two different processes need to write to the same table, so I need to do something to release the write lock from one before the other picks it up. Thoughts?

logos-electromechanical, 2016-07-14:
Maybe we can just all the close function release() and re-implement it as something that maintains the connection but drops the write lock when and if we switch to another SQL.

logos-electromechanical, 2016-07-16:
Ok, figured it out, finally. The problem is that there are several places where a SELECT statement is prepared and sent to the DB but not run all the way to completion, for whatever reason. That locks the DB file against further INSERTs until the statement in question is deleted or reset. So I added sqlite3_reset() calls in the appropriate places and now it appears to work. I will close this once I have a chance to ring it out more thoroughly.

logos-electromechanical, 2016-07-20:
Ok, I think that solved it. There's still some instances of various different processes stepping on each other WRT the database, so I need to add some re-tries at some point.

closed 2016-07-20 by logos-electromechanical